Open-Pectus/Open-Pectus #11: Sentry for error logging

nneskildsf, 2023-01-12:
Use [sentry](https://docs.sentry.io/platforms/python/) to log errors.

nnsivll, 2024-01-25:
2 sentry instances - one for errors in the pectus python code and one for errors in the UOD

mpo-mjolner, 2024-05-08:
We have 3 components that we should monitor
* Aggregator
* Engine
* Frontend

Decisions
* Use one or multiple Sentry Projects?
  * Preferably one project, "openpectus" with components seperated by some value. Alternatively one sentry project for each component.
* Log traces? This is quite heavy, especially because by default it traces FastAPI rest and websocket calls. This should probably be turned off except when perfornance testing. It is likely to affect performance considerably.
* Messages. We probably want to trace log messages of level Info and up. We also want to be able to change the level for individual modules, so that console logs and sentry logs may differ for some modules.

closed 2024-05-17 by mpo-mjolner