Open-Pectus/Open-Pectus #22: Design node identity and value equality

mpo-mjolner, 2023-02-07:
It seems, we'll need both value equality and identity equality. Value equality is possibly an expensive operation because a node value ultimately depends on the whole tree. Identity equality (the python default) is cheap but depending on node object lifetimes, serialization and mutation, it may not be sufficient. This must be analyzed in depth for each of the use cases.

Use cases:
- Interpretation: During interpretation, the tree is navigated, which requires many node comparisons. This should be fast. At this time the tree is static so identity equality seems appropriate. Tag values do change during interpretation but the tree structure does not.
- Editing: When editing the tree can be heavily mutated. This indicates value comparison in order to detect all possible changes.
- Unit testing: Value comparison is required to avoid testing stale objects

nneskildsf, 2025-05-14:
Be careful adding `__eq__` methods to the `Node` class as this will also require us to implement `__has__`.

mpo-mjolner, 2025-08-29:
This has been implemented/resolved.

We use both identity and value comparison. Node id's are sourced on the client (editor) which delivers each code line with a uuid.