Open-Pectus/Open-Pectus #25: Define system tags and commands

mpo-mjolner, 2023-03-27:
Define the built-in tags and commands that are available independently of any specific UOD

Tags
* Base
* Run counter
* Block time
* Run time
* Process time
* ...?

Commands
* Increment run counter
* Stop
* Pause
* Hold
* ...

mpo-mjolner, 2023-03-27:
See also #19

nneskildsf, 2023-04-13:
Tags:
* Connection Status:
  `Connected`, `Disconnected`
* Method Status:
  `OK`, `Error`
* System State
  `Run`, `Paused`, `Hold`, `Wait`, `Stopped`. If `Paused` and `Held` simultaneously, then `Paused` takes priority.
* Clock:
   Seconds since epoch
* Run Time:
   Set to 00:00:00 when user starts method. Run timer increments when `System State` is not `Stopped`.
* Process Time
   Set to 00:00:00 when user starts method. Process time increments when `System State` is `Run`.
* Block Time
  Set to 00:00:00 when leaving a block due to `End block` or `End blocks`. Block time increments when `System State` is `Run`.
* Accumulated Volume
  Set to 0.00 L when user starts method.
* Block Volume:
   Set to 0.00 L when leaving a block due to `End block` or `End blocks`.
* Accumulated CV
  Set to 0.00 CV when user starts method.
* Block CV
  Set to 0.00 CV when leaving a block due to `End block` or `End blocks`.
* Mark
   Holds value assigned by `Mark`-instruction. Value is reset to empty string at time of saving current values to disk.
* Base:
  `L`, `h`, `min`, `s`, `mL`, `CV`,  `g`, `kg` etc.
* Batch:
   Hold value assigned by `Batch`-instruction.
* Run Counter:
  Integer value which can be assigned by `Run counter` instruction or incremented by 1 by `Increment run counter` instruction.

nneskildsf, 2023-04-13:
Instructions:
* Base
* Block
* Watch
* Alarm
* Simulate
* Simulate off
* Mark
* End block
* End blocks
* Pause
* Hold
* Wait
* Stop
* Restart
* Error
* Warning
* Info
* Batch
* Increment run counter
* Run counter

mpo-mjolner, 2024-10-07:
Current status:

Instructions missing:
* Batch
* Info, Warning, Error
* Simulate, Simulate off

Tags missing:
* Batch

Discuss Accumulated Volume, Block Volume, Accumulated CV, Block CV. These are implemented but possibly not quite correctly regarding how accumulation should work when exiting blocks.

nneskildsf, 2025-02-09:
* Batch command issue: #342
* Info, warning, error are implemented #466 
* Simulate: #340

I have checked implementation of accumulator tags in `lang.exec.tags_impl` and they look good to me.

Closing this since progress is tracked in the above mentioned issues.

closed 2025-02-09 by nneskildsf