Open-Pectus/Open-Pectus #26: Complete interpreter commands

mpo-mjolner, 2023-03-27:
Add state
* Status: Started, Completed, Cancelled, Pending?
* Progress/ commands with duration (needs clarification)
* Temporal field? Totalization (needs clarification)
* Times of start, completion, cancellation?

Relates to #19 and #25

nneskildsf, 2023-04-13:
Hi

### Status
I agree with the states `Started`, `Completed`, `Cancelled` and `Pending`.

I think we should be precise about `Pending` though. I would consider an instruction to be `Pending` if it was issued manually during `Hold` or `Paused` execution state. I would however *not* consider it to be `Pending` if it is in the Method but has not yet been reached by the interpreter.

### Progress, temporal field and values at change of status
Some instructions may have a limited lifespan. Take for instance a ramp from a current value `u0` to a new value `u1` during the period `dT`. Then the calculation in the instruction internals would be something like `u = u0 + (u1-u0)/dT * (t-t0)` where `t0` was the time the instruction was `Started` and `t` is the current time.
Other commands might have a duration which is defined by the value of a monotonically increasing tag such as a volumetric totalizer. In that case the equivalent formula is `u = u0 + (u1-u0)/dV * (V-V0)`.

To ensure generality I think it would be optimal to take a snapshot of the entire tag space with their values at the time of status change to `Started` and attach that copy to each instruction object. That way, any instruction can access the value of any tag at the point of change of status to `Started`, and use it as it sees fit for progress calculation. This would definitely be my recommendation.

If you agree, then I think we should use the same approach for the other `Status` changes as well.

mpo-mjolner, 2023-10-17:
Related #216

closed 2023-10-17 by mpo-mjolner