Open-Pectus/Open-Pectus #48: Investigate Azure container options

mpo-mjolner, 2023-04-24:
Github directly supports deployment to Azure Web App and Azure Functions App.

## See this for comparison between ACS and Azure web apps
https://stackoverflow.com/questions/47232013/difference-between-azure-container-service-and-web-app-for-containers#47233235

## See this for comparison between ACI and ACA
https://serverfault.com/questions/1083358/difference-between-azure-container-instances-and-azure-container-apps

## Bigger MS comparison
https://learn.microsoft.com/en-us/azure/container-apps/compare-options
## From this, it appears Azure App Service or Azure Container Instance suit our needs best. 

As always, the Azure naming is ambigous and it is not immidiately clear whether Azure Web App and Azure App Service is the same thing. Looking into it, they are in fact the same thing (https://stackoverflow.com/questions/66460146/what-is-the-difference-between-azure-web-app-and-azure-app-service).

The immidiate conclusion is thus that we can use Azure Web App/Azure App Service for hosting Aggregator and Githib supports deploying to it.

nneskildsf, 2023-04-24:
Hi. I will re-open this ticket because I think the scope is wider than Github and Azure possibilities. I think the scope should also include what is possible in the target enterprise system. I will report on this in the morning meetings when I have made progress.

nneskildsf, 2023-07-03:
I have investigated the following:
1. Deployment to Azure Container Instance
2. Deployment to Azure Web App
3. Deployment to Azure Virtual Machine

I think option 1 is optimal for testing and development and that option 3 is optimal for production deployment.

Currently, an Azure Container Instance is running at http://github.openpectus.org/ for development purposes. A virtual machine on the closed enterprise network is also operational for "production" use.

Br Eskild

closed 2024-06-06 by mpo-mjolner