Open-Pectus/Open-Pectus #110: Lifecycle of engine in aggregator

nneskildsf, 2023-06-15:
The Aggregator runs on a centralized server and dispatches interactions between Engines and End Users accessing the GUI.
Engine instances initiate their connection to the Aggregator with a registration request. The Aggregator receives this request and the Engine becomes available in the GUI.

This process raises a number of questions. What happens if the engine is restarted? What if the unit operation that an Engine represents is changed at some point?
In other words, what is the lifecycle of an engine in the aggregator?

The registration request can possibly offer the following:
1. Unit Operation Definition (UOD) that the engine is executing
2. Name of Unit Operation defined in the UOD
3. Computer name of computer executing the Engine 
4. (Time of registration)

<details><summary>Code sample regarding computer name</summary>

```python
import socket
socket.gethostname()
```
</details>

Assumptions:
1. The Aggregator creates a record in the database of the Engine based on the information above.
2. **One** computer can only execute **one** instance of an Engine executing **one** UOD

## Process for registration of Engines in Aggregator
1. Aggregator is running
3. Engine is launched
4. Engine sends registration request to Aggregator with UOD, name of Unit Operation and computer name.
6. Aggregator handles request. Outcome depends on the information sent by the engine as follows:
   1. If a record of the name of unit operation *and* computer name **are not** currently in database of Engines known to the aggregator: A new record is made in the database and the Engine is listed in the UI.
   2. If a record of the unit operation **does exist**, although not running on the same computer name, then this new engine is handled as above: A new record is made in the database and the Engine is listed in the UI.
    3. If a record of the name of unit operation *and* computer name **are** currently in database of Engines known to the aggregator: 
        1. Verify that assumption 2 is met by checking that the engine record does not have a live connection. If not, then decline registration and log the error.
        2. Set a flag on the existing record making it "Inactive" or otherwise outdated. The intention is to only show "Active" or up-to-date Engines in the GUI
        4. Create a new record for this request which is marked active and online.

## Handling of Engine shut down
The WebSocket between Engine and Aggregator will be terminated when the Engine shuts down. The Aggregator should set a flag marking the Engine record as "Not online".

## Historical data
Historical data exists in the context of the Engine session in which they came to be. The Engine session executes a UOD which defines this context. Hence it is important that "inactive" or outdated Engine sessions are not removed, since they hold the definitions necessary to produce plots etc.

jhk-mjolner, 2023-06-21:
I'm not sure why I'm assigned to this, @nneskildsf ?

nneskildsf, 2023-06-21:
> I'm not sure why I'm assigned to this, @nneskildsf ?

Fixed :-). Br Eskild

mpo-mjolner, 2023-07-03:
#138 is related, the secret must be part of the registration protocol

jhk-mjolner, 2023-09-27:
@nneskildsf have I understood correctly that the only reason to provide the UOD to the Aggregator when registering is to be able to get the plot configuration for historical data? 
If so, perhaps we should only provide the information necessary for plot configuration when registering, and not the entire UOD, which currently is a complex class in the Engine, which is not easily serialized and not something the Aggregator needs to know about (I think). 
And I suggest pulling this feature (providing plot configuration data to the Aggregator at time of registering) out into it's own issue, as it doesn't seem necessary or fully clarified right now.

jhk-mjolner, 2023-09-27:
@nneskildsf There's also been talk about providing a hash of the UOD and/or the UOD file modified time to the Aggregator when registering, to be able to detect if the UOD has changed since last registering. I'm not sure if this is an desired feature as it's not listed in this issue, and I'm not sure we want to reject an engine trying to register, because it's UOD has changed since last time. Do we?
If we do want this, it could possibly also be its own issue, to allow for more time for clarification.

mpo-mjolner, 2023-09-29:
> @nneskildsf have I understood correctly that the only reason to provide the UOD to the Aggregator when registering is to be able to get the plot configuration for historical data? 

There are other reasons, so I think it is ok as is.

jhk-mjolner, 2023-10-04:
I have split things that are either not clarified enough yet, or prioritized for later, or simply not implemented by the branch for this issue, into other issues.
@mpo-mjolner do we agree that the requirements listed in this issue is either captured by another issue, or implemented sufficiently by the branch, or do we need more issues/implementation before the branch can be merged and close this issue?

closed 2023-10-09 by jhk-mjolner