Open-Pectus/Open-Pectus #138: Define secret in UOD

nneskildsf, 2023-06-28:
Define a secret in the Aggregator that a UOD must know/contain for an engine to be able to establish a connection.

Please create clear and concise error messages that engines with a wrong secret will get if they attempt to connect.

This measure should ensure that random engines cannot connect to an aggregator on an open network.

Br Eskild

jhk-mjolner, 2024-06-06:
We need to conside exactly what the Threat Model is and what attacks and attackers we want to protect from.
Adding a secret by itself isn't a difficult change in the code, but introduces some complexity wrt. how the secret is distributed to the UODs/engines.

nneskildsf, 2024-11-16:
My assumption is that the aggregator will always run in a segregated network. As such, we don't need to protect from attackers but just from people who don't know too well about the consequences of their actions.

The main "threat" that I anticipate is that team A operates an aggregator for their own use, which team B attempts to use without first consulting with team A. In this case, the purpose of the secret is that team B has to communicate with team A to agree that they can be allowed use team A's aggregator before they get access (get the secret).

I propose we implement this by requiring basic authentication for the "register engine" endpoint on the aggregator. See https://fastapi.tiangolo.com/advanced/security/http-basic-auth/#simple-http-basic-auth for technical details.
It's a simple solution to implement. The credentials can be configured in environment variables.

Br Eskild

closed 2025-02-12 by mpo-mjolner