Open-Pectus/Open-Pectus #163: Protocol congestion strategy

mpo-mjolner, 2023-09-01:
The current Engine-Aggregator protocol implementation is subject to congestion in case of network problems. We should consider ways to avoid this. 

The main problem is that in case Engine cannot deliver TagsUpdated messages, they will quickly fill up local buffers and possibly crash the Engine. A solution would be to watch for problems and back off on delivery in problem situations. This will require api changes e.g. add a 'changed_since' parameter across the call stack.

nneskildsf, 2025-02-25:
I will create a test case with 10 engines running on one aggregator.

nneskildsf, 2025-05-17:
Hi

I have used `test_startup_connection.py` as a starting point and [developed a performance test case](https://gist.github.com/nneskildsf/d9a65d65e2c7fe81e640461a5c2e3f1b). I can run 50 engines (with active runs) simultaneously and get about 4 % CPU use and 0.3 mb/s disk activity for the aggregator process on Windows 11 Intel i5-13600 32 GB ram.

I don't think that it makes sense to run this test in our test suite but it's definitely nice to run periodically.

The result indicates to me that we don't have performance problems regarding engine/aggregator.
I guess I can use Playwright to simulate a lot of active LSP sessions? I will look into that at some point.

Br Eskild

eskildsf, 2025-05-31:
It turns out that we had a bug causing the message buffer to grow very quickly when the connection is interrupted. That is fixed in #785 so now an engine only produces about 2 messages every 5 seconds.

The reconnect method waits for a randomized amount of time thus ensuring that engines don't attempt to reconnect at exactly the same time.

I've made an attempt to measure a realistic (but heavy) cycle of messages https://gist.github.com/eskildsf/10a560b36f535f65b3ed37029c67cba1. The result is that a cycle of messages to add to the buffer would weigh in at 88040 bytes. During 24 hours we would thus accumulate `88040 * 24*60*60/5 = 1451 MB` of messages. The hourly yield is 60 MB.

This is a large number but it also assumes that the running method actually runs uninterrupted for the entire duration of the aggregator outage. I don't think this will happen in practice. 4 hours of data is more reasonable but that is still 240 MB.

The question is then what rate of messages the aggregator can handle. I assume this is limited by disk performance and network speed. In any case, it might take a while to transmit the messages to the aggregator once it is back online.

Right now we block everything to transmit messages once we are reconnected. I think we can do better by transmitting the message buffer in a concurrent task and continue buffering messages until transmission is done.

closed 2025-06-30 by mpo-mjolner