Open-Pectus/Open-Pectus #201: Support installation for "regular user"

jhk-mjolner, 2023-10-09:
- Write instructions/README.md for a regular user to get set up and running with an aggregator running either locally or cloud and a number of engines running on a number of machines.
- Make a choice of whether we want to support this via PIP installing a package, or running a (or several) docker image(s), or using PyInstaller.

@nneskildsf What kind of "regular user" are we targeting? What is their experience with Python and pip installations? Could they have an older (or newer) python version than we require, and could installation of pectus in their global environment give them problems?

PyInstaller
- Support download of single .exe (and instructions?).
- Support placing UOD as prenamed file in same directory?
- Support AD integration and other configuration via .ini file?

PIP Install:
- Support being PIP installed in a global, (or conda?), (or venv?) environment.

Docker Images:
- Publish either a single docker image, or one for aggregator and one for engine, that can easily be run.

nneskildsf, 2023-10-10:
Thank you for the initiative. This is important.

I think it is reasonable to support pip installing as well as providing a docker image as we currently do. I don't think we should provide a PyInstaller package.

Basically we should support users in development as well as in production. For development purposes (especially when developing UODs) I think it is helpful to be able to pip install engine and aggregator. For production purposes I expect the aggregator to be run in a Docker container and that engines are pip installed.
That's why I don't think we need to provide a docker image "just" for the engine component.

I don't understand the distinction between pip installing globally or in a venv? I guess both work the same.

Br Eskild

nneskildsf, 2025-01-29:
Package is released to pypi: https://pypi.org/project/openpectus/
Docker image is published: https://github.com/Open-Pectus/Open-Pectus/pkgs/container/open-pectus
GUI program is released: https://github.com/Open-Pectus/Engine-Manager-GUI/releases/tag/release

closed 2025-01-29 by nneskildsf