Open-Pectus/Open-Pectus #207: Updating engine while running

nneskildsf, 2023-10-11:
Consider the case that an engine is running a long experiment. In this case it should be possible to update the engine code, stop the engine and start it again from the state it was in when it stopped.

nneskildsf, 2025-02-09:
Proposal:

1. Implement `__getstate__` and `__setstate__` methods on `Engine` class and its dependencies to support `pickle` serialization.
2. Save pickle when stopping engine process.
3. When starting the engine with a UOD for which a pickle is available, peek into it to check if a run was in progress. If so, load engine state from disk.
4. Implement flag on `pectus-engine` to force fresh start without consideration of a pickle.

I have this sort-of-working. It doesn't look too bad.

mpo-mjolner, 2026-10-29:
This is possible using the crash-recovery mechanism. Crashing the engine using Ctrl-Break would save the current engine state and make it possible to resume from that point in time.

We could consider renaming "crash recovery" to "snapshot" maybe...