Open-Pectus/Open-Pectus #282: Implement OPCUA hardware

mpo-mjolner, 2024-01-16:


mpo-mjolner, 2024-01-16:
@nneskildsf Så er der en branch klar. Det er nok nødvendigt at du får sat et udviklingsmiljø op som beskrevet i https://github.com/Open-Pectus/Open-Pectus/blob/main/DEVELOPMENT.md

nneskildsf, 2024-01-17:
Hi

I realized that we need the following functionality, and we didn't touch on it in our discussion yesterday.

When we start an engine then it should immediately write safe output values to the hardware endpoints. This is necessary to bring the hardware into a known state.
I think it would be sensible for this logic to be a part of the engine initialization which would trigger a write to all hardware endpoints. As such I will not implement this logic into the hardware-module itself.

What do you think?

mpo-mjolner, 2024-01-19:
> Hi
>
> I realized that we need the following functionality, and we didn't touch on it in our discussion yesterday.
>
> When we start an engine then it should immediately write safe output values to the [...]

This makes good sense. I have create issue #286 for this feature.

mpo-mjolner, 2024-01-19:
@nneskildsf The code looks very good. It is more involved than I thought, so I would like a talk about it. Go ahead and make a pull request if you're happy with it, then we can walk through it.

closed 2024-01-30 by mpo-mjolner