Open-Pectus/Open-Pectus #323: Log errors for frontend

jhk-mjolner, 2024-02-13:
https://github.com/Open-Pectus/Open-Pectus/issues/311

UOD author could make an error in the UOD.
Erroneous command in method editor.

Maybe create another issue for when we have a hardware layer ready for logging/error handling

jhk-mjolner, 2024-08-05:
Wrt. to
> Log statements are unique. (1000 repeats of the same error are not useful)

I think we need to know the exact situation where this can occur, so we can fix the issue correctly.
If it e.g. is a case of not reporting the same error from hardware each tick, we should probably handle it near the hardware, rather than filter just before sending to frontend. 
Filtering on "same text" just before delivering to frontend could hide some repeat errors with significant information, such as clicking a command button twice, where filtering could hide that it failed both times.

jhk-mjolner, 2024-08-05:
A potential solution could be to deduplicate concecutive lines with the same text, keeping the latest timestamp, and counting how many repeats there is, like the chrome debugger does.

mpo-mjolner, 2024-08-13:
Note: The errors sent to aggregator by engine are not just text messages forwarded to the frontend. They are  part of aggregator's model state (contained in RunData which is contained in EngineData). Incoming error log entries are handled in aggregator.from_engine.error_log_changed which currently appends the entries.

mpo-mjolner, 2024-08-14:
Created a spike on aggregation in branch https://github.com/Open-Pectus/Open-Pectus/tree/323-log-errors-for-frontend
Missing persistence changes to save the new entries in db

mpo-mjolner, 2024-10-07:
Review spike, merge it in and create an issue for frontend changes

mpo-mjolner, 2024-10-08:
Hi @jhk-mjolner , this branch is ready for the frontend changes regarding duplicated log entries

closed 2024-10-09 by jhk-mjolner