Open-Pectus/Open-Pectus #327: "All Process Values"-window

nneskildsf, 2024-02-19:
Currently, the Process Unit page shows a selection of process values. I propose that we add a button/link to a page, for example as a pop-up windows, showing all process values grouped by hardware layer I/O direction and sorted by name.
This will aid debugging and increase transparency into the engine state.

jhk-mjolner, 2024-02-20:
So the only difference between this new view and the current view of the selection of process values is the amount of values and a grouping by "hardware layer I/O direction".

Some questions:
- Could this be done in the existing view with a toggle?
- Could the same grouping, or a grouping by another parameter, be relevant for the normal selection of process values?
- I expect the new view should also be live updated with values, right? Or would a one-time snapshot be fine/preferred?
- How many process values will "all process values" typically be?
- I believe (correct me if I'm wrong) not every tag has a matching hardware register, and therefore some will have no i/o direction. Should these just go under a category "none" or something?

nneskildsf, 2024-02-20:
Thank you for the questions @jhk-mjolner. My best attempt at some answers are below:

> So the only difference between this new view and the current view of the selection of process values is the amount of values and a grouping by "hardware layer I/O direction"?
> 
> Some questions:
> 
> * Could this be done in the existing view with a toggle?

I think that is possible. Whatever maximizes clarity and overview.

> * Could the same grouping, or a grouping by another parameter, be relevant for the normal selection of process values?

I think the normal section should be defined by the UOD author who might have put some thought into the specific order of process values.

> * I expect the new view should also be live updated with values, right? Or would a one-time snapshot be fine/preferred?

Yes, it should definitely update live.

> * How many process values will "all process values" typically be?''

Between 20 and 100 would be my best estimate. That's also why grouping and alphabetical ordering is essential.

> * I believe (correct me if I'm wrong) not every tag has a matching hardware register, and therefore some will have no i/o direction. Should these just go under a category "none" or something?
Yes, exactly. The categories are: Input, Output, Input & output (for tags that are both inputs and outputs) and "N/A" for remaining process values.

closed 2024-02-28 by jhk-mjolner