Open-Pectus/Open-Pectus #340: Add simulation instructions Simulate and Simulate off

mpo-mjolner, 2024-03-21:


mpo-mjolner, 2024-04-15:
Docs from v1:

Simulate:

# Stop simulation of measured tag.

# Example 1
Simulate: UV01 Absorbance = 1 AU
Wait: 5s
Simulate off: UV01 Absorbance

mpo-mjolner, 2024-04-15:
@nneskildsf What does `Absorbance` mean in the example above? If the tag name is `UV01`, it does not seem necessary, technically speaking.

mpo-mjolner, 2024-04-15:
Add a way to express that a tag is simulated so that frontend can display it differently.

nneskildsf, 2025-02-11:
Adding @jhk-mjolner because we need to update the frontend for this issue. We need a way to indicate that a process value is simulated. I figure we can do this by setting a blue background color or perhaps a thin border on the process value.

jhk-mjolner, 2025-02-27:
Is it indicated in the data sent to frontend that a tag is simulated, and I just need to visualise it? Or is more still needed before I can do the frontend part?

nneskildsf, 2025-02-27:
> Is it indicated in the data sent to frontend that a tag is simulated, and I just need to visualise it? Or is more still needed before I can do the frontend part?

Hi. It is not currently indicated in the data sent to the frontend. That is a part of this task. I think @mpo-mjolner will let you know when this task is ready for you.

jhk-mjolner, 2025-02-27:
Okay. Assign me again when it's ready for me.

mpo-mjolner, 2025-02-27:
It is going to take a while before I have simulation data ready.

Go ahead and add a boolean field 'simulated' on TagValue (router dto and protocol). It should default to false. If we can easily avoid sending the default value, that would be my preference.

Assign me back so I can fill in the value when its ready.

jhk-mjolner, 2025-02-28:
>If we can easily avoid sending the default value, that would be my preference.  

I'm not entirely sure what you mean by this. Do you mean it should omit the value instead of sending `simulated: false` in json?

mpo-mjolner, 2025-02-28:
> > If we can easily avoid sending the default value, that would be my preference.
> 
> [@mpo-mjolner](https://github.com/mpo-mjolner) I'm not entirely sure what you mean by this. Do you mean it should omit the value instead of sending `simulated: false` in json?

IIRC you mentioned something about skipping defaults values when we upgraded pydantic. So if that is something we sometimes do already, doing it here too would be preferable.