Open-Pectus/Open-Pectus #344: Discuss block concurrency

mpo-mjolner, 2024-04-02:


mpo-mjolner, 2024-04-02:
This method indicates that block concurrency can occur in v1. It fails at runtime with the error "Multiple Blocks on the stack. This is wrong.".

Base: s
Watch: Run Counter < 2
    Block: B
        8.0 Mark: B1
        16.0 Mark: B2
        24.0 Mark: B2
        End block

Block: A
    10.0 Mark: A1
    20.0 Mark: A2
    30.0 Mark: A2
    End block

Mark: C

mpo-mjolner, 2024-04-04:
Conclusion:
* Watch and Alarm commands run concurrently via an interrupt mechanism, but their bodies do not. Once a Watch or Alarm is activated and its condition is true, the command body should be excuted as if that code was placed at the instruction pointer's position.

nneskildsf, 2025-02-25:
Consider `Block` in `Macro`.

closed 2025-10-27 by mpo-mjolner