Open-Pectus/Open-Pectus #352: In commands Wait, Hold and Pause, allow implicit use of base unit in duration

mpo-mjolner, 2024-04-11:
In then commands Wait, Hold and Pause, a duration can be specified:
`Wait: 5s`.

Expand this notion to allow leaving out the unit and inplicitly use the current base unit:
```
Base: L
Wait: 5
```
This would mean "wait for 5L".

mpo-mjolner, 2024-04-11:
This could be extended to also allow specifying a non-time unit, as long as that unit is a valid base unit:
```
# Assume X is defined as a base unit using a totalizer in the UOD
Wait: 5X
```

This would mean "wait until the totalizer defined for X reaches a value of 5.0"

nneskildsf, 2025-02-06:
Hi. Thank you for suggesting this. I think it makes sense but I think we should prefer units to be assigned explicitly. This increases clarity and makes it more safe to copy-past and move around P-code segments.