Open-Pectus/Open-Pectus #359: Fix process value units

mpo-mjolner, 2024-04-22:


mpo-mjolner, 2024-04-22:
We should look into the display of units. This screenshot shows a 'null' units for time. In this case no unit should be shown. 

[image]

The data delivered at by `/api/process_unit/MIAWLT-1645-MPO_DemoUod/process_values` for `Category` and `Time` is:
```
  {
        "name": "Category",
        "value": "Rising",
        "value_unit": null,
        "value_type": "string",
        "commands": [],
        "direction": "input"
    },
    {
        "name": "Time",
        "value": 20.019849061965942,
        "value_unit": null,
        "value_type": "float",
        "commands": [],
        "direction": "input"
    },
```

We should probably also document this better in the backend source. E.g. it is not obvious whether `value_unit` should be the empty string or `None` for no unit.

jhk-mjolner, 2024-06-26:
wrt. the documentation, I think `None` is more correct when there is no unit for a given value and will be easier to detect where relevant.
But both null and empty string will show up the same in the frontend now.

closed 2024-06-26 by jhk-mjolner