Open-Pectus/Open-Pectus #423: Add configuration options for https and wss

mpo-mjolner, 2024-08-13:
We may need to be able to configure the url scheme of the various connections::
- Engine http(s) connection to aggregator (in EngineDispatcher)
- Engine ws(s) connection to aggregator (in EngineDispatcher)
- Aggregator (uvicorn startup in AggregatorDispatcher)
- Frontend http(s) connection to aggregator (in ?)
- Frontend ws(s) connection to aggregator (in ?)

It is possible that SSL termination is done by a seperate component (web proxy) rather than Uvicorn. This may affect connections.

mpo-mjolner, 2024-09-03:
Aggregator: 
Just need to add ssl options to uvicorn.run:
ssl_keyfile="./localhost+4-key.pem",
ssl_certfile="./localhost+4.pem"
This should allow frontend to access using https.

mpo-mjolner, 2024-10-15:
@jhk-mjolner  got engine and frontend access working with the https://github.openpectus.org deployment using
```
pectus-engine.exe -ahn github.openpectus.org -ap 443
```
while using the patched EngineDispatcher (setting self._post_url, aggregator_health_url and rpc_url)

It now makes sense to add a "secure" option to EngineDispatcher that sets these options.

So something like:
pectus-engine.exe -ahn github.openpectus.org -ap 443 -secure

jhk-mjolner, 2024-10-15:
Could we maybe entirely remove the option of manually configuring a port number, and always use 80 for the aggregator and 443 or 80 for engine, depending on the -secure flag? It could potentially avoid a lot of problems with mismatching port configurations in the future. 

Of course, if it's a requirement that you can control the port, then we must support it. But could we in that case have a single source of truth for the port number, somehow? I find it confusing having multiple defaults set in multiple places.