Open-Pectus/Open-Pectus #446: Check implemented units

mpo-mjolner, 2024-09-03:
These are the current units
```
QUANTITY_UNIT_MAP = {
    'time': ['s', 'min', 'h', 'ms'],            # SI quantities
    'length': ['m', 'cm'],
    'mass': ['kg', 'g'],
    'temperature': ['degC', 'degF', 'degK'],
    'amount_of_substance': ['mol'],
    'volume': ['L', 'mL'],                      # Derived quantities
    'flow': ['L/h', 'L/min', 'L/d'],
    'pressure': ['Pa', 'bar', 'pascal'],        # Note: pint prefers pascal over Pa so we define both.
    'mass flow rate': ['kg/h', 'g/s', 'g/min', 'g/h'],
    'percentage': ['%'],                        # Custom quantity but supported by pint
    'column volume': ['CV']                     # Custom quantity
}
```

mpo-mjolner, 2024-09-03:
The units 'AU' and 'mS/cm' will be added

nnsivll, 2024-09-10:
Please also include Hz

closed 2024-09-16 by mpo-mjolner