Open-Pectus/Open-Pectus #450: CSV file format frontend/backend

mpo-mjolner, 2024-09-09:


mpo-mjolner, 2024-11-29:
To generate a backend archive file, start a run and stop it again. The file name is available in the tag Archive filename while it is running. Then compare the file to the frontend report. I think the backend archive file is missing the "header" data at least. If there are other differences we can discuss how to resolve them.

jhk-mjolner, 2024-12-04:
Extra data printed in engine archiver, not printed (as columns) in aggregator:
```
Datetime (UTC),
Base,
Run Counter,
Block,
Block Time [s],
Process Time [s],
Clock [s],
Method Status,
Connection Status,
Run Id,
Mark
```

Extra data printed in aggregator csv file, not printed in engine archiver:
More or less everything in the metadata section, except run id and start (and end) times.

If we were to introduce a metadata section of the engine archiver file, it would be potentially troublesome to write the end time and contributors list, as they are only finalized after the run. There's also information currently only available in the aggregator, and pushing that information down to the engine is a complexity.

In general: I don't think these files should aim to contain the same information, and it is my impression that @nneskildsf agrees with this. Aggregator csv is for further analysis, and engine archiver file is for debugging/forensic purposes when something goes wrong and all information is not available in the aggregator file.