Open-Pectus/Open-Pectus #485: Commands registered in UOD do not show up in frontend Commands pane on Unit Operation page

nneskildsf, 2024-11-07:
Hi

Commands registered in `demo_uod.py` do not appear in the frontend Commands pane on the Unit Operation page.

Br Eskild

mpo-mjolner, 2024-11-19:
We will do this in two iterations:
1)  Implement the basic functionality and autogenerate example documentation
2) Consider improvements once step 1 in deployed. This allows us to test on all our oud examples and decide what changes to make. Examples could be
- Add headers or tabs in the frontend to make it easy to distinguish Uod commands from System commands
- Add a documentation field in uod commands that enable the uod author to describe each command

mpo-mjolner, 2024-11-19:
Step 2 is #501

mpo-mjolner, 2024-11-19:
Implemented the basics. Now we can try it out and consider improvements.

closed 2024-11-19 by mpo-mjolner