Open-Pectus/Open-Pectus #496: Change uod command function signature for long running commands

mpo-mjolner, 2024-11-14:
This issue proposes a change to how uod command functions are defined. It proposes to support generator functions instead of the current use of `cmd.set_complete()`. It is mostly a style change but allows internal improvements because internal engine commands already generator functions.

_Details:_
In command execution function, we currently require the call `cmd.set_complete()` in all execution functions to indicate when the command is complete. This allows the engine to support long running uod commands. An example is this from demouod:
```
    def reset(cmd: UodCommand, **kvargs) -> None:
        count = cmd.get_iteration_count()
        max_ticks = 100
        if count == 0:
            cmd.context.tags.get("Reset").set_value("Reset", time())
            cmd.context.hwl.reset_FT01()  # type: ignore
        elif count <= max_ticks:
            progress = count/max_ticks
            cmd.set_progress(progress)
        else:
            cmd.context.tags.get("Reset").set_value("N/A", time())
            cmd.set_complete()
```
Note that if the `set_complete()` call if forgotten, the command will run indefinitely.

In the proposed approach we remove `set_complete()` and instead allow command execution functions to be a generator function . Then the `yield` can be used until the command is complete: This could look like this:
```
    def reset(cmd: UodCommand, **kvargs):
        max_ticks = 100
        for iteration in range(max_ticks):
          if iteration == 0:
            cmd.context.tags.get("Reset").set_value("Reset", time())
            cmd.context.hwl.reset_FT01()  # type: ignore
            yield
        elif count <= max_ticks:
            progress = count/max_ticks
            cmd.set_progress(progress)
            yield
        else:
            cmd.context.tags.get("Reset").set_value("N/A", time())      
```
Commands that complete in one iteration, just return when try're done and do not need to use `set_complete()` or `yield`. This eliminates the possibility of the uod author forgetting the `set_complete()` call.

nneskildsf, 2024-11-17:
Great idea. This makes a lot of sense. You are very welcome to implement this.

Br Eskild

mpo-mjolner, 2024-11-18:
Nice. We need to consider when because it is a breaking change in uod definitions (all command exec functions need to change a bit)

closed 2025-02-25 by nneskildsf