Open-Pectus/Open-Pectus #498: Engine should be able to resume if stopped and started again

mpo-mjolner, 2024-11-15:
@nneskildsf you mentioned this feature. We should discuss how it should work as some point.

nneskildsf, 2024-11-17:
Thank you for creating the issue.

Consider the following cases that all start the same way:
An engine registers itself and is ready. A user starts a run with the engine, which runs for a while. Then...

Case A:
After a while the network connection between the engine and the aggregator is interrupted. This could be due to power cycle of a network component or unplugging and reconnecting a network cable - even a driver update to a network interface.
The engine detects that the WebSocket to the aggregator is severed. It buffers time-series data that it would otherwise have sent to the aggregator and continues executing the current method and any long-running commands.
The aggregator also detects the severed WebSocket. The engine state is marked as "Unavailable" in the frontend. It is not possible to submit commands or make changes to the method while the engine is "Unavailable".
Shortly afterwards connection is possible again, so the engine reconnects to the aggregator and continues running as usual.

Case B:
After a while the network connection between the engine and the aggregator is interrupted. The interruption exceeds a threshold `TIME_TO_RUN_WITHOUT_CONNECTION` (a good default value might be 1 minute).
The engine detects the severed WebSocket and keeps on running as in Case A. Since the connection has not been re-established within the threshold timeframe it stops the running process. The engine stops in exactly the same way as if the user had initiated the stop in the frontend.
The aggregator also detects the severed WebSocket. At first the engine is marked as "Unavailable" as in Case A. After the threshold is exceeded it is marked as "Not online". Whatever data is available from the run is saved in "Recent runs" as if the run had been stopped by a user.

With this in place we can confidently handle most Engine/Aggregator interruptions gracefully. I hope this is specific enough to be implementable. I am happy to supply additional details.

Br Eskild

mpo-mjolner, 2024-11-18:
Thanks @nneskildsf , this is very helpful.

Case A is implemented in full. Most of Case B is also implemented, but I think some details are missing.

FYI, the error handling documentation is [here](https://github.com/Open-Pectus/Open-Pectus/blob/main/ERROR_RECOVERY.md)

I got the impression that there was another case regarding the engine being interrupted (shutdown/crash) while in a run. We don't currently support any recovery in this case.

nneskildsf, 2024-11-18:
Thank you for reminding me.

Case A and B represent usecases where the aggregator and engine are functional.

Let's define another case, Case C, as well which covers a condition where the engine fails. I think this is more relevant than cases where the aggregator fails because the engine involves user-supplied code. The aggregator **should** be stable and I assume it is. Otherwise we will find out thanks to Sentry.

Case C starts the same as A and B.

An engine registers itself and is ready. A user starts a run with the engine, which runs for a while. Then the engine stops unexpectedly. This could be due to a Python exception or simply due to a user pressing ALT+F4 or End Task in task manager.
In this case I want to be able to start the engine again and continue exactly where I left off. The entire state of the engine should be loaded (runlog, running commands, method, logged time series data). The only exception is that the engine should be in the `Paused` state such that a user must click a button in the frontend to acknowledge the intent to continue.

nneskildsf, 2025-02-06:
This issue is related to https://github.com/Open-Pectus/Open-Pectus/issues/207.