Open-Pectus/Open-Pectus #514: Fix issue with pcode in docstring

mpo-mjolner, 2024-11-28:


mpo-mjolner, 2024-11-28:
@nneskildsf A general question about running/interpreting uod commands has come up: When running a method, we don't wait for uod commands to complete before moving on to the next instruction. This has the effect that we actually consider the method to be completed as soon as we run out of code lines, even if one (or more) uod commands are still running. Is this the intended behavior?

And by extension, if we wait for uod commands before a method is complete the same question could be asked for blocks - should a Block wait for any uod commands to finish before it completes?

We don't have any other means to wait for uod commands except that the Stop and Restart commands specifically pauses and waits for uod commands.

It is probably not important in the current case of running pcode in docstrings - the current behavior just means that we may accept commands that have not yet completed - but in a more general sense it might be important.

nneskildsf, 2024-11-28:
The method cannot be "completed". I guess it can be exhausted but it does not have a status of its own. The run can be completed by `Stop` command or by pressing the "Stop" button. The behaviour you describe is intended.

A block is completed upon `End block`. Until that, the execution stays in the block without proceeding further. Please call me if you need clarification.

closed 2024-11-28 by mpo-mjolner