Open-Pectus/Open-Pectus #533: Choice command issue

mpo-mjolner, 2024-12-03:
There is a bug in the uod builder regarding choice commands. A process value with choices is defined using the `with_process_value_choice()` builder method. When invoked without the `command_options` argument, it expects that its tag has choices and  it should use these choices as command_options. This functionality does not work correctly.

Related functionality:
When fixing this, we might also add functionality to generate example code for choice commands in addition to what we already have for regex based commands (#524).

Additionally, we should refactor the processing of commands sent from the frontend, so that for choice commands, we accept verbatim pcode instead of accepting a named option and correlating that with readings. This makes the processing simpler and makes it so that generated example command test the same input command processing as used when accepting commands from the frontend.

mpo-mjolner, 2024-12-03:
Related: We currently don't support multiple commands (process values with entiher entry or choices) based on the same tag. Support for this could be added by explicitly naming the process values rather than just using ther tag names implicitly.

nneskildsf, 2025-02-10:
> Related functionality: When fixing this, we might also add functionality to generate example code for choice commands in addition to what we already have for regex based commands ([#524](https://github.com/Open-Pectus/Open-Pectus/issues/524)).

Fixed in #596.

nneskildsf, 2025-05-16:
I don't think we should couple tags and commands. They have to stay separate to avoid complexity.

I suggest we don't try to create `command_options` from `tag.choices` but instead require `command_options` to be specified.

closed 2025-06-26 by mpo-mjolner