Open-Pectus/Open-Pectus #577: Editor method state

mpo-mjolner, 2025-01-07:
State in editor should be changed
- Look-ahead must be added
- Do not lock lines until the next instruction has actually started
- ...

jhk-mjolner, 2025-07-28:
We already have merging logic in the frontend to handle when lines with unsaved edits have been run by the engine (and therefore locked), so I guess we can do something similar to merge edits made by the engine back into the frontend.

One question though: when the edit fails on the engine side, is the solution not always to revert to the pre-edit state? Could we not just do that every time? Do we need the interim edited-and-failing state? 
When will the fail happen? If it happens during the edit operation we could probably abort/undo the edit. But if it happens in later ticks it would be more complicated.

I'm curious to understand the use-cases where the engine needs to edit the method.

mpo-mjolner, 2025-08-07:
@jhk-mjolner Thanks for input. I'll try to explain the use case.

The user makes an edit and clicks Save
* If the change is valid, it is accepted by engine, which possibly adds a few changes of its own (line breaks after certain blocks that didn't have it already). Success and the updated method is returned to the frontend which should update the editor.
* Else, the Save operation is aborted by engine and an error is returned to the frontend and the user.  The editor should still contains the user's changes and the user can make corrections and try saving again or click 'Revert' to revert the changes back to the engine's version. The reason for edit failure should just be a single line of error text which in most cases just repeats an error already shown by LSP. Other error cases are 'race conditions', e.g. if an edited line has been started in the mean time.

It looks like we can have a synchronous flow such that the frontend POST to set method waits for the engine result and returns it (Currently this is not the case and errors may occur later - this will be corrected). 

Note: It does not really matter whether the method is currently running or not. There are just more error cases if the method is running.

mpo-mjolner, 2026-08-21:
@MarkMjolner 

Currently, MethodState that describes the method for the frontend looks like this:
```
class MethodState(Dto):
    started_line_ids: list[str]
    executed_line_ids: list[str]
    injected_line_ids: list[str]
```

We should extend this to include properties `locked_line_ids` and `semi_locked_line_ids`:
- The current frontend behavior is to detect locked lines using is_started and is_executed. This should now just use locked_line_ids directly
- Semi locked lines are locked, except that it is allowed to add a line break at the end of the line

Examples of semi-locked lines:

```
Block:
    Mark: A
```
The Mark: A line will always be semi-locked, regardless of whether it is executed or not. This is because the Block never ends when there is no End block.

mpo-mjolner, 2026-08-21:
The backend implements getting the method state in MethodManager._get_method_state()

jhk-mjolner, 2026-08-27:
Just a note: I believe (haven't looked through the code) the frontend uses separate started and executed line ids so it can highlight the currently running line (started, not executed) differently that executed lines. If started and executed line ids are turned into one list `locked_line_ids`, then that sparate highlighting would also be lost. Are we sure we want to disard that feature?

MarkMjolner, 2026-08-27:
@jhk-mjolner I think the idea is to keep started and executed as they are, so we can still use them for highlighting, and then introduce locked_line_ids as a separate field. So we essentially just decouple locking behavior from the started and executed fields :)

closed 2026-09-15 by MarkMjolner