Open-Pectus/Open-Pectus #609: Automatic update of version on push to main

nneskildsf, 2025-01-10:
Reference: https://stackoverflow.com/questions/54310050/how-to-version-build-artifacts-using-github-actions

mpo-mjolner, 2025-01-14:
hardcode version "0.1.dev"
replace during build

mpo-mjolner, 2025-01-14:
@nneskildsf : I think the version patching works now. But the pypi upload still fails, now with this:
```
Uploading distributions to https://upload.pypi.org/legacy/
Uploading openpectus-0.1.905.tar.gz
WARNING  Error during upload. Retry with the --verbose option for more details. 
ERROR    HTTPError: 400 Bad Request from https://upload.pypi.org/legacy/        
         Can't have direct dependency: reloading@                               
         https://github.com/nneskildsf/reloading/archive/refs/heads/master.zip. 
         See https://packaging.python.org/specifications/core-metadata for more 
         information. 
```
This is annoying. We'll need another way to include `reloading`

nneskildsf, 2025-01-15:
I will fix this by removing reloading. I find it more important to publish to pypi than to use reloading.

closed 2025-01-15 by nneskildsf