Open-Pectus/Open-Pectus #627: State leaks through uod object when executing run_example_commands during UOD validation

nneskildsf, 2025-01-20:
I observed the following error when running `pectus-engine -uod [uod_filename] -v`:

```python
Traceback (most recent call last):
  File "C:\Users\xxx\Open-Pectus\openpectus\engine\engine.py", line 629, in _execute_uod_command
    assert cmd_request.command_exec_id is not None
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\xxx\Open-Pectus\openpectus\engine\engine.py", line 388, in execute_commands
    self._execute_command(c, cmds_done)
    ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "C:\Users\xxx\Open-Pectus\openpectus\engine\engine.py", line 414, in _execute_command
    self._execute_uod_command(cmd_request, cmds_done)
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\xxx\Open-Pectus\openpectus\engine\engine.py", line 641, in _execute_uod_command
    assert cmd_request.command_exec_id is not None
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError
```

This error disappeared when I modify
```python
runner = EngineTestRunner(uod_factory=lambda: uod, pcode=example)
```
to
```python
runner = EngineTestRunner(uod_factory=lambda: copy.deepcopy(uod), pcode=example)
```
and of course add
```python
import copy
```
at the top of the file `openpectus/engine/main.py`. `copy.deepcopy` ensures that a totally new `uod` object is used in evaluation of each `EngineTestRunner`. Without `copy.deepcopy`, each `EngineTestRunner` uses the same `uod` object.

This leads me to suspect that the test cases are interfering with each other due to shared state through the `uod` object.
This is certainly a fix, but I don't know if it's the right one.

Br Eskild

nneskildsf, 2025-01-20:
It turns out that unfinished commands end up in the `uod` instance variable `command_instances`. If I clear this variable between runs then it works without `copy.deepcopy`. See the amended version of `run_example_commands` below:

```python
def run_example_commands(uod: UnitOperationDefinitionBase):
    uod.build_commands()
    logger.info("Validating UOD command examples")
    uod.hwl = NullHardware()

    def run_example_with_description(description: str, example: str) -> list[str]:
        failed_cmds: list[str] = []
        try:
            runner = EngineTestRunner(uod_factory=lambda: uod, pcode=example)
            with runner.run() as instance:
                instance.start()
                # wait up to 1 minute, that ought to be enought for everybody
                instance.run_until_event("method_end", max_ticks=10*60)
                logger.debug(instance.get_runtime_table())
                logger.debug(f"{description} executed successfully")
                instance.engine.uod.command_instances = dict()
        except Exception as ex:
            logger.error(f"{description} '{example}' failed: {str(ex)}")
            failed_cmds.append(example)
        return failed_cmds

    logger.info("Executing example commands")
    failed_cmds: list[str] = list(
      chain.from_iterable(
        map(
            lambda t: run_example_with_description(t[0], t[1]),
            uod.generate_pcode_examples()
        )
      )
    )

    if len(failed_cmds) > 0:
        logger.error(f"Example commands failed: {','.join(failed_cmds)}")
    else:
        logger.info("All example commands executed succesfully")
```

I suggest that this cleanup is added to the `openpectus/engine/engine.py` `Engine` `cleanup` method.

mpo-mjolner, 2025-01-22:
Good find. A uod instance is not really meant to live beyond the engine's lifetime. I'll fix this with one of the LSP tasks where I have encountered a similar problem.

mpo-mjolner, 2025-03-07:
This has been fixed by the engine now using InternalCommandsRegistry whose lifetime is constrained to the engine lifetime.

nneskildsf, 2025-03-17:
Unfortunately this issue persists. Problems disappear with the `copy.deepcopy(uod)` trick from the original post.

closed 2025-05-15 by mpo-mjolner