Open-Pectus/Open-Pectus #643: Run tests in parallel with pytest

mpo-mjolner, 2025-02-05:
Just run pytest -n auto - and fix a few issues than currently fail when run in parallel

nneskildsf, 2025-02-05:
@mpo-mjolner 
```console
pip install pytest pytest-xdist
pytest -n auto
```
Tests run in 48 seconds on i7 11th gen laptop.

All tests except integration tests pass reliably on my machine.
The problem in the integration tests is that both test cases try to bind the same port. Assigning different ports for the two tests fixes the issue and all tests run fine.

Br Eskild

nneskildsf, 2025-02-05:
According to the Python docs the `time.sleep(..)` function is not very accurate. The `EngineTestInstance.run_until_condition` sleeps between ticks in one shot, see the [excerpt below](https://github.com/Open-Pectus/Open-Pectus/blob/main/openpectus/test/engine/utility_methods.py#L130-L134) which is used in three occasions:

```python
last_tick_time = tick_time
elapsed = time.time() - last_tick_time
deadline = self.timing.interval - elapsed
if deadline > 0.0:
    time.sleep(deadline)
```

Replacing this with a sequence of sleeps might make the tests more reliable because variation in the actual sleep duration can be accounted for:
```python
last_tick_time = tick_time
elapsed = time.time() - last_tick_time
deadline = self.timing.interval - elapsed
if deadline > 0.0:
    while last_tick_time + self.timing.interval - time.time() > 0.0:
        time.sleep(min(last_tick_time + self.timing.interval - time.time(), 0.01))
```