Open-Pectus/Open-Pectus #670: LSP indentation

mpo-mjolner, 2025-02-19:
It would be nice to have the editor automatically indend when writing a Watch/Alarm.

Seems there is some possibilities to customize automatic indentation, but it may not be enough:
https://microsoft.github.io/monaco-editor/typedoc/interfaces/editor.IEditorOptions.html#autoIndent

mpo-mjolner, 2025-02-19:
This may also be of help:
https://microsoft.github.io/monaco-editor/typedoc/interfaces/languages.IndentationRule.html

mpo-mjolner, 2025-02-20:
Indentation is multiples of 4 spaces. Tabs are not allowed.

The commands that require increased indentation is
* Watch
* Alarm
* Block
* Macro

Out-dent
* Inside `Block`, outdent after `End block`
* Inside `Block`, outdent completely after `End blocks` (if possible)
* Inside other commands, outdent after a blank (whitespace only) line. If user outdents manually without a blank line, don't re-indent (a blank line is not required but seems a good way to guide the user)

jhk-mjolner, 2025-02-20:
> Inside other commands, outdent after a blank
I guess I could also outdent when a line doesn't start with four spaces.

Also: is a second (and further) indent level possible? Can e.g. a block be inside a watch or macro or something?

jhk-mjolner, 2025-02-20:
I have access to [this](https://code.visualstudio.com/api/language-extensions/language-configuration-guide#indentation-rules) configuration, but I'm having some problems with unindenting when there's multiple levels of indentation.
It also seems like the `decreaseIndentPattern` is tested on the new (empty) line when pressing enter, so matching an empty line would make indentation impossible.

closed 2025-02-21 by jhk-mjolner