Open-Pectus/Open-Pectus #671: Method editor symbol colors

mpo-mjolner, 2025-02-20:
We want to color symbols in the editor. Eg commands in one color and tags in another.

I expect the "symbols" call (https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#textDocument_documentSymbol) to provide the information and that the editor magically colors things

mpo-mjolner, 2025-02-24:
I will create a branch and test document so that backend returns the proper token for the test codument

mpo-mjolner, 2025-02-24:
The test result for the textDocument/documentSymbol is new merged into main. The test document is 
```
Mark
Stop
Info
```

This test result indicates that there are 3 symbols, a "Field" and two "Method"s

jhk-mjolner, 2025-02-25:
https://github.com/TypeFox/monaco-languageclient/issues/256#issuecomment-797661454 suggests we need the lsp server to report that it supports `"semanticTokensProvider"` as "full". Currently it doesn't report anything about `semanticTokensProvider` in its capabilities.
It should also response to the `textDocument/semanticTokens/full` method then, I believe.

"full" refers to taking a full document, I believe. There's also a version working on delta, and on range. I'm not sure of the details.

mpo-mjolner, 2025-03-03:
It seems the TextMate approach is preferable to symbols or semanticTokens jf #687 .

The basic pcode syntax is:
```
[threshold] command [: arguments] [# comment]
```

eg.
```
5.0 Restart # start a new iteration
```
It makes sence to give threshold, command and comment different colors, as in pectus1.

Furthermore, it seems reasonable to give different types of commands different colors, e.g. one for system commands and another for uod commands.

For Alarm and Watch commands, the argument is a triplet consisting of tag, operator and value, eg:
```
Watch:  Run Counter > 5
    Stop
```
It makes sense to give the three parts different colors as well.

We can do more but this seems a good place to start.

mpo-mjolner, 2025-03-10:
Test method:
```
1.5 Info: Starting

Watch: Run Counter > 5
    Reset
    5.0 Stop # comment

```

closed 2025-03-10 by mpo-mjolner