Open-Pectus/Open-Pectus #687: Try using a TextMate grammar for pcode

jhk-mjolner, 2025-02-27:
- Try hand-writing a pcode.tmLanguage.json file like [this](https://github.com/TypeFox/monaco-languageclient/blob/main/packages/examples/src/langium/langium-dsl/config/langium.tmLanguage.json) or [this](https://github.com/TypeFox/monaco-languageclient/blob/main/packages/examples/src/langium/statemachine/syntaxes/statemachine.tmLanguage.json).
- Then load it. See the json plugin for how.
- Can I get colors from it?

nneskildsf, 2025-02-27:
See https://github.com/Open-Pectus/Open-Pectus/blob/main/docs/pcode_lexer.py for inspiration.

jhk-mjolner, 2025-03-03:
Okay, it seems I can make a textmate grammar that colors things. So what do we want colored in which way?

I can make regexp that matches things and then 'name' them after the [TextMate convention](https://macromates.com/manual/en/language_grammars#naming_conventions). The existing themes will then color it. So I can't directly choose the colors, but that is as it should be. We just need to pick the right TextMate names for the things.

Currently I have e.g. for the control segments.
```
​{
  name: 'keyword.control.pcode',
  match: '\\b(Watch|Alarm|Block|Macro|End block|End blocks|Stop|Restart)\\b',
}
```

closed 2025-03-03 by jhk-mjolner