Open-Pectus/Open-Pectus #725: Frontend gives red card error "HTTP Error "Unauthorized" on endpoint """ on page load

nneskildsf, 2025-04-11:


jhk-mjolner, 2025-04-22:
Is this consistent every time (assuming cache not disabled), or do you have a way to consistently reproduce the error?

nneskildsf, 2025-04-24:
When I disable caching then I observe a difference in the load order. Maybe this is significant? Unfortunately I am unable to understand the TS code to identify eaxctly where the request to the `/api/recent_runs/` endpoint is made and how it differs from say the `/api/process_units` endpoint.

Br Eskild

[image]
Figure 1. Network tab in Chrome with cache disabled.

[image]
Figure 2. Network tab in Chrome with cache enabled.

nneskildsf, 2025-04-24:
Could it be that the request is made before user authentication is completed?
See this code:
```ts
export const identityInterceptor: HttpInterceptorFn = (req, next) => {
  const oidcSecurityService = inject(OidcSecurityService);
  const secureRoutes = oidcSecurityService.getConfigurations().flatMap(config => config.secureRoutes ?? []);

  if(!oidcSecurityService.authenticated().isAuthenticated) return next(req);
  if(!secureRoutes.some(secureRoute => req.url.startsWith(secureRoute))) return next(req);

  return oidcSecurityService.getIdToken().pipe(mergeMap(idToken => {
    const newReq = req.clone({
      headers: req.headers.append('X-Identity', idToken),
    });
    return next(newReq);
  }));
};
```
https://github.com/Open-Pectus/Open-Pectus/blob/main/frontend/src/app/auth/identity.interceptor.ts#L6-L19

closed 2025-05-02 by jhk-mjolner