Open-Pectus/Open-Pectus #747: Fix accessing lsp endpoint when using ng serve

mpo-mjolner, 2025-05-13:
When using `ng serve` the lsp websocket connection fails to start so lsp does not work. This can be detected right away by typing `inf` in the editor. If lsp works it should immediately suggest `Info`.

nneskildsf, 2025-05-14:
I don't know why you would use `ng serve`? Why not just run the `lsp.main` script which starts LSP+aggregator?
If there is a good reason to run *just* the LSP then I propose that I add a command line argument in `lsp.main` to accomodate this. Please let me know if this is relevant.

Br Eskild

mpo-mjolner, 2025-05-15:
Using ng serve is required for frontend development because it allows TS changes to be loaded without having to compile the entire frontend. So lsp must also work here. But I think @jhk-mjolner might be able to figure out what the problem is.

closed 2025-05-16 by jhk-mjolner