Open-Pectus/Open-Pectus #771: Webinterface as app on iOS

nneskildsf, 2025-05-20:
Add the following to `index.html`:

```html
<meta name="apple-mobile-web-app-capable" content="yes">
<link rel="apple-touch-icon" sizes="180x180" href="touch-icon-iphone.png">
```

to make the website look more like a real app when adding a shortcut to the aggregator web interface to the iOS Home Screen.

This is the best source on the topic that I could find: https://developer.apple.com/library/archive/documentation/AppleApplications/Reference/SafariWebContent/ConfiguringWebApplications/ConfiguringWebApplications.html

nneskildsf, 2025-05-20:
This is possibly also relevant: https://web.dev/learn/pwa/web-app-manifest#designing_your_pwa_experience

eskildsf, 2025-06-11:
Closed by merge of #797

closed 2025-06-11 by eskildsf