Open-Pectus/Open-Pectus #832: Make PlotColorRegion translucent

nneskildsf, 2025-08-20:
Hi.
Consider the plot excerpt below. A PlotAxis and a PlotColorRegion have the same color (blue) causing the line to be unreadable in regions where there is a background color. Is it possible to set transparency, e.g. 50% on the PlotColorRegion color? I propose that this transparency is fixed in the frontend code.

Br Eskild

[image]

jhk-mjolner, 2025-08-20:
I believe the color parameter for the colored region fully supports css colors: https://developer.mozilla.org/en-US/docs/Web/CSS/color_value
So if you want a 50% transparent color, you might be able to use something like `rgb(0 0 255 / 50%)` or `#0000ff80` if you're using a 100% blue color.

jhk-mjolner, 2025-08-20:
It should also be noted that the region is still drawn behind the line, so simply picking a brighter or different color than the line should also work. Transparent color isn't needed, but if it's the easiest way for you to get the "same" color but weaker, there's nothing wrong with it.
There's also `hsl` where you can easily brighten or desaturate the color.

closed 2025-08-20 by nneskildsf