Open-Pectus/Open-Pectus #843: Show dates in user timezone

nneskildsf, 2025-09-09:
Hi

We use GMT+0 as the timezone for all datetimes. Is it possible to show times in the users browser timezone in the frontend?

 Br Eskild

jhk-mjolner, 2025-09-09:
By default date handling in a browser is always in the browser's local timezone, so I would have expected it to already be that.
I'm guessing the user in question hasn't set their local machine or browser to GMT+0?
Or are there a particular datetime listing in the frontend that is the problem?

I can't actually test it right now as I'm getting an error at startup for the aggregator after pulling the newest from main branch:
```python
  File "C:\mjolner-kode\Novo\Open-Pectus\openpectus\aggregator\aggregator.py", line 60, in _try_restore_reconnected_engine_data
    engine_data.contributors = set(recent_engine.contributors)
                               ~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: unhashable type: 'dict'
```
Do you have an idea of what change could have caused that?

nneskildsf, 2025-09-09:
> The run log and error log looks fine to me. It's just the recent run list and details I can see being 2 hours off.
> Any other date times you've noticed being off?

No, that matches what I observe as well. Br Eskild

jhk-mjolner, 2025-09-09:
In my testing I've noticed that the run log (and probably the error log too) actually reports their times in my local GMT+2 timezone and the recent runs endpoint reports times in GMT+0.
But the actual problem is that none of them report the timezone of the datetime. That means the frontend can't correctly convert to user's local timezone. So I think we might have a regression here, where it at some point in the past it probably did report the timezone and now it doesn't. I'm not sure whether the problem is how the date times are stored in engine/aggregator or if it's perhaps a problem with how pydantic serializes it.
Do you or @mpo-mjolner have an idea of where/when this change could have happened? Or should I try to investigate blind?