Open-Pectus/Open-Pectus #846: Plot not rendering on one run of many, console reads `Error: <rect> attribute width: A negative value is not valid.`

nneskildsf, 2025-09-10:
Hi

I have conducted a number of runs using a specific equipment. All runs show up fine in the Recent Run view except one. The plot does not render at all and in the console I can see error messages, which do not appear on other runs which render fine, reading `Error: <rect> attribute width: A negative value is not valid.`.

I can transfer the sqlite database via confidential channels if desired.

Br Eskild

nneskildsf, 2025-09-15:
Hi @jhk-mjolner. I am not allowed to share files via OneDrive so that is unfortunately not an option. We use a service called BOX.NET and I have sent you an invitation to a shared folder where the sqlite database is available. Br Eskild

jhk-mjolner, 2025-09-16:
Okay, I've looked at the data and found the issue. The last value for run time is set to 0, and the frontend uses that value as its "max" for the x-axis, because any value used for the x-axis must be a monotonically rising value, so the last value should be the max value.
If I delete that line in the db, it renders the plot fine.

I'm not sure how that value appeared from the aggregator/engine. Perhaps values were read after the engine had stopped its run and reset the value to 0? So maybe it's a race condition / timing issue in the engine/aggregator? Can we put reasonable safeguards to ensure something similar doesn't happen again? These are also questions for @mpo-mjolner